roboticslab-uc3m/xgnitive #6: [cmake] Create FindECF.cmake

jgvictores, 2016-05-19:
If [ECF](http://ecf.zemris.fer.hr/) is going to be fixed as a dependency:
- [ ] Create a FindECF.cmake
- [ ] Update the CMakeLists.txt of the programs and/or libraries that use it

RaulFdzbis, 2016-05-30:
This [ commit ](https://github.com/roboticslab-uc3m/xgnitive/commit/c423079bb3cfed487a1b7250170dcf8f825e5165) implements a working FindECF.cmake file for the project.  
@jgvictores can you check it?.

RaulFdzbis, 2016-05-30:
This [commit](https://github.com/roboticslab-uc3m/xgnitive/commit/5fa433d256dd65b2f26ead37ce3efd72fae09c02) add the new library **EcfComponents** to the xgnitive repository. 

This library uses the ECF external library. The CMakeLists files were generated using the FindECF.cmake

jgvictores, 2016-05-30:
FindECF.cmake looks good!
Maybe you could edit Aquila out of the commented lines too. ;-)

RaulFdzbis, 2016-05-31:
[Done](https://github.com/roboticslab-uc3m/xgnitive/commit/84bd072ecb78657302c34020227741d705820139)!

jgvictores, 2016-06-28:
@RaulFdzbis Any reason this was reopened?

RaulFdzbis, 2016-06-28:
I dont see any. You are right, lets close it :sunglasses:

closed 2016-06-28 by RaulFdzbis