roboticslab-uc3m/xgnitive #15: Think about abbeel

RaulFdzbis, 2016-06-09:
\juansays{Think about abbeel}

jgvictores, 2016-06-22:
I've given sparse insight on this in #18. @David-Estevez Do you think this deserves a milestone?
I'd at least like to see some checkboxes to see some progress.
- [ ] 1/3
- [ ] 2/3
- [ ] 3/3

David-Estevez, 2016-06-22:
Only if you have clear in your mind what 1/3 2/3 and 3/3 mean.

jgvictores, 2016-06-24:
I believe they are rational numbers. Perhaps we could address the non-linearities of the thought process through a taylor series expansion or an unscented transform?

David-Estevez, 2016-06-27:
Due to the non-linear nature of the process an stochastic optimization of the quadrature function with an heuristic approximation by a divide-and-conquer will suffice.

jgvictores, 2016-06-27:
This is good, because another one bites the dust. However, we should also be concerned about security issues. I'd look into creating a Merkle tree to prevent tampering of the data.
cc: @smorante Could you provide any insight or ideas?

RaulFdzbis, 2016-06-27:
I didnt understood a word of what you have said @David-Estevez , so i suppose you are right. You have my support in this.

David-Estevez, 2016-06-27:
Yeah, don't stop me now, cause I'm having a good time ( having a good time).

jgvictores, 2017-11-03:
After a night at the opera, I've ticked 1/3 thanks to the DRL course which really made me work on this issue.