roboticslab-uc3m/xgnitive #18: CgdaExecution Nomenclature

RaulFdzbis, 2016-06-22:
Taking a deeper look at the architecture of the CgdaExecution program. I realized that, since CgdaExecution files (.cpp and .hpp), are the ones in charge of implementing the cost function and the evaluation. Maybe It would be better to change the name of those files to something like CgdaFitnessFuction, or something like that, that may not lead to misunderstands.

What do you think @jgvictores ?

jgvictores, 2016-06-22:
I like naming of programs to correspond to their ultimate objective as a whole. In this sense, I think that cgdaExecution is good because it describes the ultimate objective, but doesn't force a specific technique. I hope that in the future we can make the cgdaExecution implementation flexible enough as to be able to switch between approaches that use fitness functions (e.g. evolutionary computation) and others that do not necessarily need one (e.g. reinforcement learning, unless I'm terribly confused about it).
I'd see that a generic fitness function infrastructure would be good at the library layer. For now we're pretty tied to ECF, so there's no abstraction layer in XGNITIVE yet. But the specific implementations of a given functionality (e.g. in EcfComponents) kind of give a feeling of it.

@RaulFdzbis Summary: IMHO, I  would leave naming as it is.

RaulFdzbis, 2016-06-22:
Ok, you have a good point there, and i agree with the name  of the program (This means folder+execution generated file) should stay the same.

However, i was talking more about the C++ files, now named CgdaExecution.cpp and CgdaExecution.hpp. The final objective (high level objective) of those files, as you said is the CgdaExecution, but so it is for EvMono C++ files, so if we follow that nomenclature rules, at the end EvMono should be named CgdaExecution too.

jgvictores, 2016-06-22:
- I think CgdaExecution.cpp and CgdaExecution.hpp are good, because they implement the program main class.
- I agree that [cgdaExecution](https://github.com/roboticslab-uc3m/xgnitive/tree/develop/programs/cgdaExecution) EvMono.* files could be renamed CgdaFitnessFuction.* or similar.

RaulFdzbis, 2016-06-22:
Ok, thats a good option. The confussion here is that the structure/name is different. Here evMono is the main class, and CgdaExecution is the class with the fitness function. So the proposed changes are:

- CgdaExecution.* -> CgdaFitnessFunction.*

- EvMono.* -> CgdaExecution.*

Deal?

jgvictores, 2016-06-22:
My bad. Deal!

commit 819f3446 (2016-06-25): cgdaExecution and cgdaExecutionIET renamed following issue #18. This commit closes #18
closed 2016-06-25 by RaulFdzbis