roboticslab-uc3m/xgnitive #20: Add shared resources

RaulFdzbis, 2016-10-07:
In the current implementation of the xgnitive project absolute paths are used in order to find the different resources needed for the system. 

As an improvement an implementation of a shared resources folder is proposed. Something similar to what is done in [teo-main](https://github.com/roboticslab-uc3m/teo-main/blob/develop/programs/teoSim/main.cpp#L67) . This way absoulute paths would be no longer needed, increasing the portability of the project.

jgvictores, 2016-11-23:
Yes, models and configurations should go in https://github.com/roboticslab-uc3m/xgnitive/tree/develop/share
(maybe later on we will move models outside of the repository, but using the ResourceFinder is a great first step!)

jgvictores, 2016-11-23:
I'm particularly concerned about:
https://github.com/roboticslab-uc3m/xgnitive/tree/develop/programs/models
How about a teo-openrave-models repository ?

RaulFdzbis, 2016-11-23:
Yes i like the idea, as far as i know the TEO openrave models folder is used at least in teo-main and xgnitive. So maybe could be a good idea to have that folder as an independent repository.

Just a question, what do you plan to add to that repository in addition to the models?. I think that maybe having something similar to a TeoSim standalone repository with the models could be nice.

jgvictores, 2016-11-23:
My idea (related with https://github.com/roboticslab-uc3m/QA/issues/1 and https://github.com/roboticslab-uc3m/QA/issues/2) is:
- https://github.com/roboticslab-uc3m/openrave-yarp-plugins <- as is, with more capabilities (camera..)
- https://github.com/roboticslab-uc3m/teo-openrave-models <- new
- https://github.com/roboticslab-uc3m/teo-main <- lightweight, binds the previous two among others

RaulFdzbis, 2016-11-23:
That's look fine to me. The only thing that still unclears is what would you put in the new teo-openrave-models repository.

jgvictores, 2016-11-23:
Only the teo openrave models.

RaulFdzbis, 2016-11-23:
Ok that can work, lets do it.

RaulFdzbis, 2016-11-24:
Done! https://github.com/roboticslab-uc3m/teo-openrave-models

jgvictores, 2016-11-24:
Cool,  looks good! Thanks!

RaulFdzbis, 2016-11-24:
@jgvictores should we not keep this open until we configure xgnitive to work with the teo-openrave-models?

jgvictores, 2016-11-24:
Okay!

RaulFdzbis, 2017-03-07:
@jgvictores  Should do we close this with the last commit?

jgvictores, 2017-03-07:
Still have to move things around, such as:
https://github.com/roboticslab-uc3m/xgnitive/tree/develop/programs/cgdaExecutionOET/conf