roboticslab-uc3m/xgnitive #23: Move datasets (experiments folder) out of repository

jgvictores, 2016-11-21:
Move datasets (experiments folder) out of repository.
On other occasions, we have published on https://sourceforge.net/projects/roboticslab/files/Datasets/ but now could think of more modern places such as [ResearchGate](https://www.researchgate.net/profile/Juan_Victores) (which allows publishing datasets and provides DOI) or [zenodo](https://zenodo.org) (must check advantages).

RaulFdzbis, 2016-11-23:
The experiments database has been moved to: https://zenodo.org/record/168156#.WDVvLpNVKlM.

jgvictores, 2016-11-23:
Thanks @RaulFdzbis !

1. Could you move multipleExecutionsBatch.sh to programs, to erase the experiments folder completely?

2. The dataset name seems very general. Let's leave this issue open until we specify what it actually contains, e.g. specific paper (and what it will contain in the future if it can be modified).

RaulFdzbis, 2016-11-23:
Done!

jgvictores, 2016-11-23:
Thanks @RaulFdzbis !

closed 2016-11-23 by jgvictores