roboticslab-uc3m/xgnitive #25: Document programs, libraries and apps

jgvictores, 2016-11-29:
The contents should match in quantity:
- [ ] https://github.com/roboticslab-uc3m/xgnitive/tree/develop/programs vs http://robots.uc3m.es/dox-xgnitive/group__xgnitive__programs.html
- [ ] https://github.com/roboticslab-uc3m/xgnitive/tree/develop/libraries vs http://robots.uc3m.es/dox-xgnitive/group__xgnitive__libraries.html
- [ ] https://github.com/roboticslab-uc3m/xgnitive/tree/issue-22-yarpmanager-xml/share vs http://robots.uc3m.es/dox-xgnitive/group__xgnitive__applications.html

RaulFdzbis, 2016-12-02:
Finished in 88c1179a709f312b187e5b00a30bf14f99449ed2 , should we close it? Or let it open to start documenting classes, functions...?

jgvictores, 2016-12-02:
@RaulFdzbis It's looking really good, I'm closing the issue, thanks!!!

closed 2016-12-02 by jgvictores