roboticslab-uc3m/xgnitive #32: Compatibility issue between demonstration-feature-selection and dtw package

RaulFdzbis, 2017-08-10:
Due to changes in the dtw python package, there is a compatibility problem between the code used in [demonstration-feature-selection](https://github.com/roboticslab-uc3m/xgnitive/tree/2f1f6436a32f653681e53a8731c26f9d901047db/demonstration-feature-selection/src) and the last version of [dtw](https://pypi.python.org/pypi/dtw/1.2).

The problem comes from the definition of the dtw function, in the current code the function is called with two parameters (the two time series to compute), while in the last versions of dtw the function is defined using three parameters (two time series and a dist parameter that defines the norm).

**The current proposed fix to this problem is to use the 1.0 version of the [dtw](https://pypi.python.org/pypi/dtw/1.0) library while the code is not updated.**

jgvictores, 2017-08-10:
Are we using a `requirements.txt`?
It is nice to use (in addition to install guides) because you can force a version. Tiny example [here](https://github.com/UC3Music/songbook-tools/blob/develop/requirements.txt).

jgvictores, 2017-08-10:
PS: Then just `sudo pip install -r requirements.txt`.

RaulFdzbis, 2017-08-11:
Simple and effective, just how i like it :D. Added in 64e0076a1fd957907296a64bde41499575f7ea2d , thanks @jgvictores!!

jgvictores, 2017-08-22:
I guess we should update [doc/xgnitive-install.md](https://github.com/roboticslab-uc3m/xgnitive/blob/85c92d3d08cca70a8d5e45050742fdaad84f8f4b/doc/xgnitive-install.md)?

RaulFdzbis, 2017-08-23:
Since right now is only required in the demonstration-feature-selection branch, I only updated the [doc/xgnitive-install.md](https://github.com/roboticslab-uc3m/xgnitive/blob/8c5f6fa6300333a8fb9a2b06b0adf042fe084348/doc/xgnitive-install.md) at that branch.

jgvictores, 2017-08-23:
Oh, Ok! Hadn't noticed that. Thank you!

jgvictores, 2018-03-14:
Yep!

closed 2018-03-15 by RaulFdzbis