roboticslab-uc3m/xgnitive #39: Openrave compatibility issues with rllab environment on conda

RaulFdzbis, 2017-10-17:
So when I try to build Openrave inside the rllab environment generated using conda i get the following error:

```bash
-- Compiling OpenRAVE Version 0.9.0, soversion=0.9
-- Using cmake version 3.5.1
-- Using double precision
-- detected system processor: x86_64
  File "<string>", line 1
    import numpy; print numpy.get_include()
                            ^
SyntaxError: invalid syntax
  File "<string>", line 1
    from distutils.sysconfig import get_python_lib; from myrelpath import relpath; print relpath(get_python_lib(1,prefix='/usr/local'),'/usr/local')
                                                                                               ^
SyntaxError: invalid syntax
-- failed to get python site-package directories via get_python_lib
-- installing to /usr/local, python exe /home/raul/anaconda2/envs/rllab3/bin/python, python subdir is , program suffix is '0.9'
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   regex
--   filesystem
--   system
--   python
--   thread
--   iostreams
--   date_time
--   chrono
--   atomic
-- found boost version: 105400
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Could NOT find GMP
-- Could NOT find MPFR
-- MPFI needs GMP and MPFR
-- Using local crlibm
-- compiling with double extended (long double) floating point
CMake Warning at CMakeLists.txt:479 (find_package):
  By not providing "FindASSIMP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ASSIMP", but
  CMake did not find one.

  Could not find a package configuration file provided by "ASSIMP" with any
  of the following names:

    ASSIMPConfig.cmake
    assimp-config.cmake

  Add the installation prefix of "ASSIMP" to CMAKE_PREFIX_PATH or set
  "ASSIMP_DIR" to a directory containing one of the above files.  If "ASSIMP"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Checking for module 'assimp'
--   No package 'assimp' found
CMake Warning (dev) at modules-cmake/CheckCXXSourceRuns.cmake:16 (IF):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "IS_X86_64" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  CMakeLists.txt:505 (check_cxx_source_runs)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- WARNING: you are using the obsolete 'PKGCONFIG' macro, use FindPkgConfig
-- libxml2 found
CMake Warning at CMakeLists.txt:599 (find_package):
  Could not find a package configuration file provided by "COLLADA_DOM"
  (requested version 2.3) with any of the following names:

    COLLADA_DOMConfig.cmake
    collada_dom-config.cmake

  Add the installation prefix of "COLLADA_DOM" to CMAKE_PREFIX_PATH or set
  "COLLADA_DOM_DIR" to a directory containing one of the above files.  If
  "COLLADA_DOM" provides a separate development package or SDK, be sure it
  has been installed.

-- Checking for module 'minizip'
--   No package 'minizip' found
-- Compiling with COLLADA support from sources!
-- compiling local ann library
CMake Warning at CMakeLists.txt:710 (find_package):
  By not providing "FindFPARSER.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "FPARSER", but
  CMake did not find one.

  Could not find a package configuration file provided by "FPARSER"
  (requested version 4.4.3) with any of the following names:

    FPARSERConfig.cmake
    fparser-config.cmake

  Add the installation prefix of "FPARSER" to CMAKE_PREFIX_PATH or set
  "FPARSER_DIR" to a directory containing one of the above files.  If
  "FPARSER" provides a separate development package or SDK, be sure it has
  been installed.

-- Checking for one of the modules 'fparser'
-- compiling local fparser library
-- Checking for module 'liblog4cxx'
--   No package 'liblog4cxx' found
-- compiling local convexdecomposition library
-- Geometry Parsing: Collada DOM 
-- Geometry Parsing: Using ivcon
-- extracting models to /home/raul/repos/openrave/src
-- MATLAB installation not found, is mex in the system path?
-- Octave installation not found, is 'mkoctfile' in the system path?
  File "<string>", line 1
    import sympy; print sympy.__version__
                            ^
SyntaxError: invalid syntax
-- failed to find python sympy system installation
-- System sympy (v=) is not right version, using local sympy
CMake Error at CMakeLists.txt:811 (install):
  install DIRECTORY given no DESTINATION!

CMake Warning (dev) at modules-cmake/CheckCXXSourceRuns.cmake:16 (IF):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "HAVE_ALL_PYTHON_HEADERS" will no longer be
  dereferenced when the policy is set to NEW.  Since the policy is not set
  the OLD behavior will be used.
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:13 (check_cxx_source_runs)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- python and boost-python found
CMake Error at python/CMakeLists.txt:7 (install):
  install FILES given no DESTINATION!

-- Checking for module 'bullet'
--   No package 'bullet' found
-- Could not find Bullet Physics/Collision engine. Please install ver 2.80 from http://www.bulletphysics.com
-- FindMoby: Could not find EventDrivenSimulator.h
-- Could not find Moby Physics engine. Please install from https://github.com/PositronicsLab/Moby 
-- enabling FFMPEG support
-- boost regex found
-- Could NOT find OpenSceneGraph: Found unsuitable version "3.2.0", but required is at least "3.4" (found /usr/lib/libosgDB.so;/usr/lib/libosgQt.so;/usr/lib/libosgGA.so;/usr/lib/libosgFX.so;/usr/lib/libosgText.so;/usr/lib/libosgViewer.so;/usr/lib/libosgManipulator.so;/usr/lib/libosg.so;/usr/lib/libOpenThreads.so)
-- Could not find OpenSceneGraph v3+. Please install OSG (http://www.openscenegraph.org/projects/osg)
-- Detected SoQt/Coin3D GUI, making plugin
-- Detected ODE multi-threading extensions, but not sure if should enable them... define ODE_USE_MULTITHREAD variable to use them
-- ODE found, building oderave plugin, precision=DOUBLE
-- Configuring incomplete, errors occurred!
See also "/home/raul/repos/openrave/build_rllab/CMakeFiles/CMakeOutput.log".
See also "/home/raul/repos/openrave/build_rllab/CMakeFiles/CMakeError.log".
```
The command I use for the build is:

```bash
cmake .. -DOPENRAVE_PLUGIN_FCLRAVE=OFF
```

My initial guess is that the problem comes because **rllab is using the python 3.5.2 version which is not compatible with openrave**, see:

```bash
    import numpy; print numpy.get_include()
                            ^
SyntaxError: invalid syntax
```

This problem comes because in python 3.5 the syntax of print have been changed from **print X**  to **print(X)**.

One **solution** can be to have a double installation of the python 2.7 version and python 3.5 inside the rllab environment. However, I am not sure how to do this yet.

jgvictores, 2017-10-17:
Total wild guess, but it may even be the print function, which changed from Python 2 to 3...
Which led me to https://github.com/rdiankov/openrave/issues/523 open issue asking for python 3 support. 😱😱😱

jgvictores, 2017-10-17:
> Total wild guess, but it may even be the print function, which changed from Python 2 to 3...
Which led me to https://github.com/rdiankov/openrave/issues/523 open issue asking for python 3 support. 😱😱😱

Apparently my `totally wild guess` matches your assumtion too. :joy:

> One solution can be to have a double installation of the python 2.7 version and python 3.5 inside the rllab environment. However, I am not sure how to do this yet.

Not sure if this would be good... I'd expect they are using some Python 3 features at least. Maybe it'll be quicker to roll YARP in after all...

PeterBowman, 2017-10-17:
You probably know this, but pasting anyway (from https://github.com/rll/rllab):

> rllab only officially supports Python 3.5+. For an older snapshot of rllab sitting on Python 2, please use the py2 branch.

You can still use some of Python 3 features on 2.7 ([SO](https://stackoverflow.com/q/7075082)).

jgvictores, 2017-10-17:
@PeterBowman I think we're going to continue with the component-based approach via YARP which was an option we had already contemplated, but thanks a lot for the info!

RaulFdzbis, 2017-10-18:
So since we have decided to use YARP instead of openrave + conda, right now solving this problem is no relevant for the developing of xgnitive, so i will close this issue, until a solution is found, or it is relevant for this project again.

closed 2017-10-18 by RaulFdzbis