roboticslab-uc3m/xgnitive #40: Conda dont find openravepy

RaulFdzbis, 2017-10-18:
So i tried to reinstalled openravepy after having conda installed, however conda still cant find openravepy inside its python path. This means, this error appear:

```bash
import openravepy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named openravepy
```

Doing some research i found this related [issue](http://openrave-users-list.185357.n3.nabble.com/ImportError-No-module-named-openravepy-tried-installing-openrave-multiple-times-td3711211.html).

David-Estevez, 2017-10-18:
I imagine you already know this and might not be the issue,  but just in case....

Conda usually creates its own Python environment (as it is a good Python programming practice).  Packages that you had installed outside this environment won't be found, and new packages that you want to be recognized by packages within Conda must be installed within this environment or added to the environment's  PYTHONPATH.

RaulFdzbis, 2017-10-18:
yeah thanks @David-Estevez, thats the reason why i reinstalled openrave after installing conda, but it still does not work. Maybe i need to create another environment or maybe activate the root conda environment (however i suppose that this root environment is activated by default).

David-Estevez, 2017-10-18:
Activating the Conda environment in the terminal you are installing openrave should suffice. But you are right, that environment is typically  activated by default after installing Conda.

[Internet](https://conda.io/docs/user-guide/tasks/manage-environments.html) says that you can check the different environments with:
`conda info --envs`

RaulFdzbis, 2017-10-18:
The temporal solution found thanks to @jgvictores  to this problem was to copy the contents from  "/usr/local/lib/python2.7/site-packages/openravepy/" to  "/anaconda2/lib/python2.7/site-packages/openravepy".

Anyway we have decided to avoid using openrave with conda, and use YARP instead.

Aldo thanks for the help @David-Estevez !!! :D

closed 2017-10-18 by RaulFdzbis