roboticslab-uc3m/xgnitive #41: python swigg error with YARP: Wrong number or type of arguments for overloaded function

RaulFdzbis, 2017-10-19:
So the problem comes, when i try to do this in order to define a parameter for the yarp device:

```bash
mentalOptions= yarp.Property
mentalOptions.put("device", "remote_controlboard") #device
```

The error i got is the following:

```bash
  File "/home/raul/repos/rllab/scripts/run_experiment_lite.py", line 137, in <module>
    run_experiment(sys.argv)
  File "/home/raul/repos/rllab/scripts/run_experiment_lite.py", line 121, in run_experiment
    method_call(variant_data)
  File "grid_3D_yarp.py", line 10, in run_task
    env = normalize(GridWorld3DEnvYarp())
  File "/home/raul/repos/xgnitive/programs/drl/envs/grid_world_3D_env_yarp.py", line 78, in __init__
    mentalOptions.put("device", "remote_controlboard") #device
  File "/home/raul/anaconda2/envs/rllab3/lib/python3.5/site-packages/yarp.py", line 2545, in put
    return _yarp.Property_put(self, *args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'Property_put'.
  Possible C/C++ prototypes are:
    yarp::os::Property::put(yarp::os::ConstString const &,yarp::os::ConstString const &)
    yarp::os::Property::put(yarp::os::ConstString const &,yarp::os::Value const &)
    yarp::os::Property::put(yarp::os::ConstString const &,yarp::os::Value *)
    yarp::os::Property::put(yarp::os::ConstString const &,int)
    yarp::os::Property::put(yarp::os::ConstString const &,double)
```
It looks like the c++ yarp.property function is complaining about the arguments passed to it, however in python "xxxx" is defined as a ConstString, so i suppose is a swigg problem.

David-Estevez, 2017-10-19:
Could it be that you are missing some parenthesis in the object instantiation?

Try:
```
mentalOptions= yarp.Property()
```

RaulFdzbis, 2017-10-19:
Omg yeah, In my head I thought i had added that parenthesis :D

Thank you @David-Estevez, that solved the problem!

closed 2017-10-19 by RaulFdzbis

David-Estevez, 2017-10-19:
You wouldn't believe the amount of times I've done the exact same thing and spent hours looking for the error... :smile:

jgvictores, 2017-10-19:
@David-Estevez The error message is actually quite cryptic. Thanks a lot for helping out!