roboticslab-uc3m/xgnitive #42: /openraveYarpPaintSquares/rpc:s unknown command

RaulFdzbis, 2017-10-23:
The problem comes when trying to send a command to "/openraveYarpPaintSquares/rpc:s" to get the state of the painted wall. When doing this, this port responds with an empty bottle and does not recognise the command. The code I am using is:

```bash
        remotePaint = "/openraveYarpPaintSquares/rpc:s"
        localPaint = "/cgda/openraveYarpPaintSquares/rpc:c"

        self.rpcClient=yarp.RpcClient()
        self.rpcClient.open(localPaint) #Connect to local paint

        while True: ##do-while
            yarp.Network.connect(localPaint,remotePaint)
            print("Wait to connect to paint server...\n")
            time.sleep(self.yarpDelay)
            if self.rpcClient.getOutputCount() != 0:
                break
        print ("Paint server available. \n")

        #Read state
        self.cmd= yarp.Bottle()
        self.res= yarp.Bottle()
        cmd= "reset"
        print(cmd)
        self.rpcClient.write(self.cmd,self.res)

        print("The size is: ", self.res.size())

        for i in range(self.res.size()):
            print(self.res.get(i).asInt())
            print(i)

        cmd= "get"
        self.rpcClient.write(self.cmd,self.res)

        print("The size is: ", self.res.size())

        for i in range(self.res.size()):
            print(self.res.get(i).asInt())
            print(i)
```

The output of **yarp read /read tcp+log.in://openraveYarpPaintSquares/rpc:s** is:

```bashrc
yarp read /read tcp+log.in://openraveYarpPaintSquares/rpc:s
yarp: Port /read active at tcp://163.117.150.58:10040/
yarp: Receiving input from /openraveYarpPaintSquares/rpc:s to /read using tcp+log.in
[rpc] () ("unknown command")
[rpc] () ("unknown command")
```

David-Estevez, 2017-10-23:
Could it be that you are sending empty bottles?

You are creating an empty bottle
`self.cmd= yarp.Bottle()`
an then asigning a string to another different variable (without `self`)
`cmd= "reset"`
to finally send the empty bottle
`self.rpcClient.write(self.cmd,self.res)`

I think you should put the "reset" string within the created bottle (I guess the command is something similar to `self.cmd.put(string)`, but I'm not sure).

jgvictores, 2017-10-23:
Also, use `yarp rpc /portName` for debugging RPC ports.  :-)

jgvictores, 2017-10-23:
Oh, regarding the comment of @David-Estevez that's also right, but the correct syntax is more in the line of `self.cmd.addString("reset")`.

RaulFdzbis, 2017-10-23:
"Hi my name is Raul and i am new at python"

Yes that missing "self." was the problem ;D, thank you to both of you!

closed 2017-10-23 by RaulFdzbis