roboticslab-uc3m/xgnitive #43: yarp rpc /teoSim/rightArm/rpc:i "get encs" responsing with complex format

RaulFdzbis, 2017-10-25:
When trying to **get the state of the encoders** from TEO, the output format is something like this.

```bash
yarp rpc /teoSim/rightArm/rpc:i
>>get encs
Response: [is] encs (30.0 0.0 0.0 0.0 0.0 0.0) [tsta] 70 1508927958.17612 [ok]
```
When trying to parse that string in python to get the state of each encoders things are not so straightforward as i think should be. With the following code:

```
for i in range(self.res.size()):
    state_now.append(self.res.get(i).asDouble())
print("The state now is: ",state_now)
```

I get the following **output**:

```
The state now is:  [29545.0, 0.0, 0.0, 1635021684.0, 89.0, 1508928750.360022, 27503.0]
```

I think there is a problem with the formatting of the ouput signal. To **solve** this I did the following:

```
response=self.res.toString()
response = response.replace("(", "") #Delete parenthesis from the string
response=response.split(" ") #split string
print("response2: ",response[2]) 
```
With the **output**:

```
response2:  30.0
```

**So I think that a possible improvement for the future could be to change the format of the output of "get encs" to have a more proper format.**

jgvictores, 2017-10-25:
Better use the `IEncoders` interface: [see example](https://github.com/roboticslab-uc3m/asibot-main/blob/c975cd5b6e049be9a59e1c11b807faefe51fee6f/example/python/testRemoteRaveBot.py#L45-L68) interface.

jgvictores, 2017-10-25:
PS: Parsing `[is] encs (30.0 0.0 0.0 0.0 0.0 0.0) [tsta] 70 1508927958.17612 [ok]` would be a bit more like (note that syntax may be slightly different, do not have `ipython` right now):
```python
b = get(2).asList()
```
and then doing similar to yours:
```
for i in range(b.size()):
    state_now.append(b.get(i).asDouble())
print("The state now is: ",state_now)
```

RaulFdzbis, 2017-10-25:
Thank you @jgvictores!

I used `IEncoders` as you suggested in the [code](https://github.com/roboticslab-uc3m/xgnitive/blob/d7ac379e4e8bf9f741c228a4ba39da8c1e4ffb41/programs/drl/envs/teo_paint_discrete_yarp_env.py#L318-L321).

jgvictores, 2017-10-25:
Regarding [these lines](https://github.com/roboticslab-uc3m/xgnitive/blob/d7ac379e4e8bf9f741c228a4ba39da8c1e4ffb41/programs/drl/envs/teo_paint_discrete_yarp_env.py#L323-L326) that are currently:
```
        state_now=[]
        state_now.append(float(str(v[0])))
        state_now.append(float(str(v[1])))
        state_now.append(float(str(v[3])))
````
Since `str` is just for printing, maybe they could be more efficiently implemented as:
```python
        state_now=[]
        state_now.append(v[0])
        state_now.append(v[1])
        state_now.append(v[3])
```

PS: Also careful with more `get encs` along the same code.

RaulFdzbis, 2017-10-26:
BTW: 

Confirmed this works:

>b = self.res.get(2).asList()
>
>for i in range(b.size()):
>    state_now.append(b.get(i).asDouble())
>print("The state now is: ",state_now)

Thank you @jgvictores i think we can close this issue.

closed 2017-10-26 by RaulFdzbis