roboticslab-uc3m/xgnitive #46: Allow hard-coded paint programs to work with our new friend the "parametric wall"

RaulFdzbis, 2017-12-12:
Like [here](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/5d86b9cb07be41c7112438d3d6be10da26fb2990/libraries/OpenravePlugins/OpenraveYarpPaintSquares/OpenraveYarpPaintSquares.cpp#L38).

Ps: For some programs may be a better option to let them as it be. [Example](https://github.com/roboticslab-uc3m/xgnitive/blob/0982eb342649a7ecd3a86af5e94b002be24abd43/programs/cgdaExecutionOET/cgdaExecutionOETOpenrave/CgdaPaintFitnessFunction.cpp#L9)
Ps2: This may affect other repos.

RaulFdzbis, 2017-12-12:
The main file to be changed is [OpenraveYarpPaintSquares.cpp](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/5d86b9cb07be41c7112438d3d6be10da26fb2990/libraries/OpenravePlugins/OpenraveYarpPaintSquares/OpenraveYarpPaintSquares.cpp). 

The options i have considered in order to do this are the following:

- [ ] Keep it hard-coded and change the Number of squares [here](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/5d86b9cb07be41c7112438d3d6be10da26fb2990/libraries/OpenravePlugins/OpenraveYarpPaintSquares/OpenraveYarpPaintSquares.cpp#L38). The easiest and maybe the least invasive option. The disadvantage is that we need to change the code and recompile xgnitive, anytime we want to use a different wall.

- [ ] Code a pass as param code to allow passing the Nsquares of the wall as a parameter when calling the plugin. This way we dont need to change and recompile the code anytime we change the wall. The bad thing, we need to implement code to do this and i am not sure how it will affect to all the programs using this plugin.

- [ ] Load the Nsquares  from the xml when loading the [environment](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/5d86b9cb07be41c7112438d3d6be10da26fb2990/libraries/OpenravePlugins/OpenraveYarpPaintSquares/OpenraveYarpPaintSquares.cpp#L158). Probably the best option, however i dont know if its even possible.

RaulFdzbis, 2017-12-12:
Right now I think the best option is the first one. The reason is because the advantage are quite cool, and initially we dont plan to change the Nsquares of the wall very often. The initial idea is to choose a resolution and stick with it. 

However if anyone has another opinion is welcome. If this is not the case i will try the code, and if it works the issue can be considered closed.

jgvictores, 2017-12-12:
One sec and I'll see it with you at the lab! :-)

RaulFdzbis, 2017-12-12:
Tested, and looks like it is working.

closed 2017-12-12 by RaulFdzbis

RaulFdzbis, 2017-12-13:
Since https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/635fa1017f08f7024b7b91ad32651e9c29aa0085, we can now call openraveYarpPaintSquares using params. 

[Example.](https://github.com/roboticslab-uc3m/xgnitive/blob/749bd21be1c7b2a55223072f7447d5aefd4c1c15/programs/drl/envs/teo_paint_discrete_yarp_env.py#L105)