roboticslab-uc3m/xgnitive #47: Teo Paint simulation not reaching all the squares

RaulFdzbis, 2017-12-13:
So with the new resolution of the wall, the TEO model is not able to reach all the squares in the wall. This is a screenshot of an attemp i did trying to check the reachibility of the robot.

[image]

I tried also to run the DRL algorithm and it is able to paint a few more squares, which means that is more intelligent than me, nothing new. Anyway its still no able to paint the 100% of the wall, and i think this problem will increase if we increase the resolution. So its something we have to deal.

A few options I propose are:

- [ ] Changing the position/size of the wall. Generating a smaller wall and little bit closer to the robot will help a lot. I will emphasize in the part **smaller**.
- [ ]  Changing the position of TEO in the simulation.
- [ ]  Adding more joints to the learning process, like torso and so on. This will greatly increase the reachability of the robot.

RaulFdzbis, 2017-12-13:
Right now i think that the better option is the first one, since we have a unnecessary big wall and changing the position of the wall i think its easier than changing the TEO position. Thanks to [this](https://github.com/roboticslab-uc3m/xgnitive/blob/0efe12a0d54d07311a96824bdbb1c5e17b245f60/programs/wall-creator/wallCreator.py).

 Also since adding more joints is out of the scope of our current research project, i should discard the third option. Although it should be considered for future works.

jgvictores, 2017-12-13:
Smaller is okay as long as it's not, say 50% smaller LOL.
Maybe a combination of smaller and closer wall?

RaulFdzbis, 2017-12-13:
I just remembered that the size of the wall has its meaning, it corresponds with the size of the TV in the lab. 

However, there is no need to use all the resolution, so we can maybe reduce the weidht of the wall while keeping the height, to have a nice looking application. 

Also we should be carefull about the position of the wall, maybe closer means violating other limits.

Anyway i will start working with the option 1.

RaulFdzbis, 2017-12-13:
With the last commit https://github.com/roboticslab-uc3m/xgnitive/commit/d1822719b9587130ec43fa005c8ac2405e7c199e, it looks like now it is able to reach all the wall. However, before closing the issue i will let a full experiment finish and if it works i will close the issue.

RaulFdzbis, 2017-12-14:
Working like a charm, closing this issue.

closed 2017-12-14 by RaulFdzbis

jgvictores, 2017-12-14:
Yay!