roboticslab-uc3m/xgnitive #48: Installation xgnitive dependencies guide with rllab3

RaulFdzbis, 2017-12-18:
So this weekend i was starting to install all the xgnitives dependencies in my personal computer. And since with Anaconda this process is not quite straightforward, I am doing some parallel documentation about How to install things.

My question is, where should this documentation go?, In the installation repo since is an installation guide? or in this repository?, since is a very specific case of installing things inside the rllab environment.

jgvictores, 2017-12-18:
I'd start off at `doc/xgnitive-install.md`, then we'll see how to slice and move.

RaulFdzbis, 2018-01-09:
A good idea could be to add the rllab installation instructions as a different module inside  `doc/xgnitive-install.md` rather than just add more dependencies and steps to the current installation manual.

jgvictores, 2018-01-09:
Regarding `rllab`, our current practice would be to start a `install-rllab.md` at https://github.com/roboticslab-uc3m/installation-guides

RaulFdzbis, 2018-03-15:
Since how to install rllab is well explained in the [original docs](http://rllab.readthedocs.io/en/latest/user/installation.html), i dont think the need to create a install-rllab.md.

Instead i create a `install-anaconda.md` [instruction manual](https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-anaconda.md), that i think is more tricky and I think it makes more sense to explain there how to install packages inside anaconda environments.

jgvictores, 2018-03-21:
Okay, I guess we can live with that.

So... just linking our anaconda manual, and the official rllab, from https://github.com/roboticslab-uc3m/xgnitive/blob/develop/doc/xgnitive-install.md ([currently missing](https://github.com/roboticslab-uc3m/xgnitive/blob/538422e5e1b395f9ef047fcc8699c6e09b74281b/doc/xgnitive-install.md))?