roboticslab-uc3m/xgnitive #49: How-to paintSquaresOnScreen

RaulFdzbis, 2018-01-26:
[How-to paintSquaresOnScreen](https://github.com/roboticslab-uc3m/xgnitive/blob/e5c01a3bd90cf68864fa17607137ef8479cd86d8/programs/paintSquaresOnScreen/README.md)

RaulFdzbis, 2018-03-14:
So I think this tries to say to document How-to paintSquaresOnScreen using the real tv?

@jgvictores what do you think? jajaj

jgvictores, 2018-03-15:
Yeah, the [current README](https://github.com/roboticslab-uc3m/xgnitive/blob/10b65ed9d8900fc0b06ef3bac002703fe4bf6085/programs/paintSquaresOnScreen/README.md) in `develop` is pretty empty. Just basic launch docs.

RaulFdzbis, 2018-03-15:
Some work done in https://github.com/roboticslab-uc3m/xgnitive/commit/ddc961f0edc4f85f788a4d8315e21732fa4bce66

jgvictores, 2018-03-15:
Nice, thanks!
On my behalf, it's okay to close this issue.

RaulFdzbis, 2018-03-15:
Cool!

closed 2018-03-15 by RaulFdzbis