roboticslab-uc3m/xgnitive #51: How to set termination condition for DRL/EVO experiments

RaulFdzbis, 2018-02-08:
_From @RaulFdzbis on December 14, 2017 18:26_

So the thing is how to set termination condition for DRL/EVO experiments, to have comparable results between all the algorithms.

I think the best option, to have a best comparasion between algorithms, is to let the algorithms interact for a fixed time with the simulation. Other options like limiting the number of iterations, are not valid since the definition of an iteration depends of the algorithm.

The "simplest" way to do it, i think is just by measuring the time (t) it takes to an algorithm to perform an iteration, and just perform N iteration where N*t=T for each algorithm.

**UPDATE:** The condition, paint until all the wall is painted, i dont think it will work in DRL, since in DRL we are not searching THE trajectory like in EVO, we are searching the **policy** so painting the wall one time could be a bad policy just having a lucky day. 

@jgvictores wachuduzink my bro?

_Copied from original issue: RaulFdzbis/raul-research-shop#3_

RaulFdzbis, 2018-02-08:
_From @jgvictores on December 15, 2017 14:2_

Anything left to do here, once agreed 16 squares?

RaulFdzbis, 2018-02-08:
yeah, sure, the problem still the same. How to choose the termination condition just to compare the algorithms. I am still thinking this is the best idea:

>The "simplest" way to do it, i think is just by measuring the time (t) it takes to an algorithm to perform an iteration, and just perform N iteration where N*t=T for each algorithm.

RaulFdzbis, 2018-02-08:
_From @jgvictores on December 15, 2017 14:22_

Ok!

jgvictores, 2018-12-08:
@RaulFdzbis Okay to close this?

RaulFdzbis, 2018-12-11:
Yes. Not sure if is still a problem right now. We can always reopen.

closed 2018-12-11 by RaulFdzbis