roboticslab-uc3m/xgnitive #52: Paint demonstrations

RaulFdzbis, 2018-02-08:
Paint demonstrations

RaulFdzbis, 2018-02-08:
Steps:
- [ ] launchManipulation
- [ ] gcmp (review k-d for this)
- [ ] follow recommendations of https://github.com/roboticslab-uc3m/questions-and-answers/issues/28#issuecomment-329166785

RaulFdzbis, 2018-02-08:
> gcmp (review (update, install, understand) k-d for this)

We are going to try to do k-d gcmp directly on TEO-manipulation as a first approach.

jgvictores, 2018-02-08:
Not all drivers act in torque mode with launchManipulation + gcmp. I've opened https://github.com/roboticslab-uc3m/teo-main/issues/35 to track this.

jgvictores, 2018-03-22:
Also https://github.com/roboticslab-uc3m/teo-hardware-issues/issues/14 blocks

jgvictores, 2018-11-25:
Closing as `invalid`, description was very vague regarding motivation/target.

closed 2018-11-25 by jgvictores