roboticslab-uc3m/xgnitive #59: Unify line endings

PeterBowman, 2018-07-19:
Several files use Windows (CRLF) line endings (8c2f26c9724e16a3b159816c583582f6b6a17ed5):

```bash
$ find -type f -exec file {} + | grep -i CRLF | awk '{ print $1; }' | sed -e 's/:$//'
./programs/cgdaExecutionOET/cgdaExecutionOETOpenrave/CgdaIronFitnessFunction.hpp
./programs/cgdaExecutionOET/cgdaExecutionOETOpenrave/main.cpp
./programs/cgdaExecutionOET/cgdaExecutionOETOpenrave/CgdaPaintFitnessFunction.hpp
./programs/drl/experiments/experiment_2017_10_30_11_FIRST_NO_BUG_EXPERIMENT/progress.csv
./programs/drl/experiments/experiment_2017_11_28_REMOTE-INSTREW-DYNLI-NOPEN-BIGREW-ENC/progress.csv
./programs/drl/experiments/experiment_2017_11_08_NO_DELAY_DYNAMIC_LIMITS_2/progress.csv
./programs/drl/experiments/experiment_2017_11_16_DELAY_DYNAMICLI_INSTREW_NOPENALTY/progress.csv
./programs/drl/experiments/experiment_2017_11_16_DELAY_DYNAMICLI_INCREW/progress.csv
./programs/drl/experiments/experiment_2017_11_07_NO_DELAY_DYNAMIC_LIMITS/progress.csv
./programs/drl/experiments/experiment_2017_11_15_DELAY_DINAMYCLI_INCREW/progress.csv
./programs/drl/experiments/experiment_2017_11_22_NOLOCAL-INSTREW-FIXEDLIM-NOPEN-BIGREW-ENC/progress.csv
./programs/drl/experiments/experiment_2017_11_14_DELAY_FIXED_LI/progress.csv
./programs/drl/experiments/experiment_2017_11_13_DELAY_DYNAMIC_LI/progress.csv
./programs/drl/experiments/experiment_2017_11_24_REMOTE_INSTREW-FIXEDLI-NOPEN-BIGREW-ENC/progress.csv
./programs/drl/experiments/experiment_2017_11_20_NOLOCAL_DELAY_DYNAMICLI_INTREW_NOPENALTY/progress.csv
./programs/drl/experiments/experiment_2017_11_10_DELAY_DYNAMIC_LI/progress.csv
./programs/drl/experiments/experiment_2017_11_03_FIRST_EXPERIMENT_LOCAL_OPENRAVE_N0_DELAY/progress.csv
./programs/drl/experiments/experiment_2017_12_12_REMOTE-INSTREW-DYNLI-NOPEN-64S/progress.csv
./programs/drl/experiments/experiment_2017_12_12_REMOTE-INSTREW-DYNLI-NOPEN-64S-SMALL/progress.csv
./programs/drl/experiments/experiment_2017_11_15_DELAY_DINAMYCLI_INSTCREW/progress.csv
./programs/drl/experiments/experiment_2017_10_31_NO_DELAY_FIRST_EXPERIMENT/progress.csv
./programs/drl/experiments/experiment_2017_11_23_REMOTE_INSTREW-DYNAMICLI-NOPEN-BIGREW-NOENC/progress.csv
./programs/drl/experiments/experiment_2017_11_23_REMOTE_INSTREW-DYNAMICLI-NOPEN-BIGREW-NOENC/experiment__2017_11_23_REMOTE_INSTREW-DYNAMICLI-NOPEN-BIGREW-NOENC/progress.csv
./programs/cgdaExecutionIET/cgdaExecutionIET/CgdaWaxFitnessFunction.hpp
./programs/cgdaExecutionIET/cgdaExecutionIET/CgdaIronFitnessFunction.hpp
./programs/cgdaExecutionIET/cgdaExecutionIET/main.cpp
./programs/cgdaExecutionIET/cgdaExecutionIET/CgdaPaintFitnessFunction.hpp
./programs/cgdaExecutionIET/cgdaExecutionIET/CgdaConstrainedPaintFitnessFunction.hpp
./programs/cgdaExecutionIET/cgdaExecutionIET/CgdaConstrainedWaxFitnessFunction.hpp
./programs/cgdaExecutionIET/conf/evMono_ecf_params_WAX.xml
./programs/cgdaExecutionIET/cgdaExecutionIETYarp/main.cpp
./programs/cgdaExecutionIET/cgdaExecutionIETYarp/CgdaConstrainedPaintFitnessFunction.hpp
./libraries/EcfComponents/AlgPSOInheritance.hpp
./libraries/EcfComponents/AlgConstrainedSST.cpp
./libraries/EcfComponents/AlgConstrainedSST.hpp
```

Use this:

```bash
$ find -type f -exec dos2unix {} +
$ git diff --word-diff=color --word-diff-regex=.
```

Beware of stale branches!

PeterBowman, 2018-09-04:
Thanks, @RaulFdzbis! I noticed there are two development branches of yours which might need merge+normalization as well: `demonstration-feature-selection-issue-36`, `simulated-wrinkle`.

RaulFdzbis, 2018-09-04:
Done! https://github.com/roboticslab-uc3m/xgnitive/commit/93bf9d08df0ea8429b4f934610edb4280cfff108 and https://github.com/roboticslab-uc3m/xgnitive/commit/9efc379df318d478231d7a1dcd85e11b21ac71f7.

RaulFdzbis, 2018-09-13:
Should we close this now?

PeterBowman, 2018-09-13:
I was wondering whether source files should be incidentally converted to UTF-8 (from ASCII). That might deserve a separate issue, if deemed necessary at all. Thanks, closing.

closed 2018-09-13 by PeterBowman

jgvictores, 2018-09-15:
> converted to UTF-8 (from ASCII)

AFAIK ascii is a subset of utf8. Since this is not specified explicitly in any header, if a file is fully composed by ascii characters, we can say it is also utf8. I read about this while writing https://github.com/asrob-uc3m/tutoriales/blob/master/software/character-encoding.md ([perma](https://github.com/asrob-uc3m/tutoriales/blob/c3325139502a76c40db389d0477634bcec779f7e/software/character-encoding.md)), so there should be more references there.

PeterBowman, 2018-09-15:
See also: [Should my source code be in UTF-8?](https://softwareengineering.stackexchange.com/q/152780). In the past, I did have to set UTF8 as default in Eclipse (when developing Java code) to avoid weird conversion errors.

jgvictores, 2018-09-15:
> See also: Should my source code be in UTF-8?

Nice read. Added at https://github.com/asrob-uc3m/tutoriales/commit/7c1e5cb338d56ec4161f349df2a5da48bff8d09a

> In the past, I did have to set UTF8 as default in Eclipse (when developing Java code) to avoid weird conversion errors.

Well, yes. In practice, wierd things can happen with basically any editor, and can also vary among OS. UTF8 for now works well. : )