roboticslab-uc3m/xgnitive #60: clean branches

jgvictores, 2018-12-13:
clean branches: cannot be merged automatically

RaulFdzbis, 2019-11-14:
Update: I tried to do a local merge between develop and master and worked, should i make a push to the remote?

PeterBowman, 2019-11-14:
The master branch only differs in two merge commits from develop, so you could as well either delete it and create anew or force push from current develop.

RaulFdzbis, 2019-11-14:
From master: "This branch is 2 commits ahead, 416 commits behind develop." 

That's more than two commits hahha

PeterBowman, 2019-11-14:
Nope, that's exactly two commits if your base branch is develop. Those commits are meaningless in this comparison since both come from develop anyway, thus merging develop into master means you can start a new master branch on top of develop with little or no harm (okay, you are rewritting the commit history on a shared repo, but IMO it can be assumed given how [the network graph](https://github.com/roboticslab-uc3m/xgnitive/network) for master branch looks like).

RaulFdzbis, 2019-11-15:
Ok, now I understand your point, and yeah looks fine to me to delete the master branch and create a new one in develop, that can help reduce the mess in the network :D

RaulFdzbis, 2019-11-15:
Done

PeterBowman, 2020-09-11:
Today, `develop` has been merged into `master` because of https://github.com/roboticslab-uc3m/questions-and-answers/issues/88.